renode/renode #762: Support requested to execute .robot on MacOS

spulverail, 2025-03-06:
### Description

We are trying to use Renode in our development on a STM32L452 board.
I am able to run `renode-test` with `tests/example.robot` with success but when I run it with `tests/platforms/Renesas_RA6M5.robot I get the following error and I don't know how to fix it:

```+++++ Starting test 'Renesas_RA6M5.CK IIC Board Should Work'
!!!!! Emulation's state saved to "/Users/spulverail/Documents/TechnoConcept/internship/nvp-renode//snapshots/Renesas_RA6M5.CK_IIC_Board_Should_Work.fail0.save"
!!!!! Log saved to "/Users/spulverail/Documents/TechnoConcept/internship/nvp-renode//logs/Renesas_RA6M5.CK_IIC_Board_Should_Work.fail0.log"
+++++ Finished test 'Renesas_RA6M5.CK IIC Board Should Work' in 0.09 seconds with status failed
      ╔═
      ║ InvalidOperationException: Error while opening dynamic library: dlopen(/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so, 0x0006): tried: '/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (no such file), '/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/private/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (no such file), '/private/var/folders/k7/8d9dpw_94mnbtszjm6bz6zrh0000gn/T/renode-83316/11-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))
      ║ 
      ║ 
      ║ Also teardown failed:
      ║ NullReferenceException: Object reference not set to an instance of an object
      ║ 
      ╚═
```

Can you help me understand and fix this failure please?
I am familiar to Linux and I just move on MacOS.

Thanks.

### Usage example

Here is how I am executing the robot file after activating my Python virtual-env.

For `example.robot`:
`PATH=${PATH}:/opt/homebrew/bin /Applications/Renode.app/Contents/MacOS/tests/renode-test ${PWD}/renode/tests/example.robot`

For `Renesas_RA6M5.robot`:
`PATH=${PATH}:/opt/homebrew/bin /Applications/Renode.app/Contents/MacOS/tests/renode-test ${PWD}/renode/tests/platforms/Renesas_RA6M5.robot`

### Additional information

Environment: MacBook Pro with an apple M4 pro on Sequoia 15.3.1
Renode: Either using Renode_1.15.3.dmg or using dmg generating from sources (branch: master)

LillyJinstrand, 2025-03-06:
Hi @spulverail 
This is most likely caused by using arm64 python. To run renode-test under Rosetta you need to have a x86 python virtual-env, otherwise it will try to run Renode natively instead of through Rosetta. 
Alternatively, you can try using the experimental native arm64 Renode, which is available on this branch https://github.com/renode/renode/tree/68189-aarch64-tcg-ops, build using the `--host-arch aarch64` flag to `build.sh`

spulverail, 2025-03-06:
Hi @LillyJinstrand,
Thank you for the quick answer.

I choose the second option you propose me, and now when I run the `Renesas_RA6M5.robot` test suite I get this error:

```Preparing suites
Starting suites
Running suite on Renode pid 96313 using port 49152: /Users/spulverail/Documents/TechnoConcept/internship/renode/tests/platforms/Renesas_RA6M5.robot
+++++ Starting test 'Renesas_RA6M5.Should Run Periodically Blink LED'
^CSecond signal will force exit.
!!!!! Emulation's state saved to "/Users/spulverail/Documents/TechnoConcept/internship/nvp-renode//snapshots/Renesas_RA6M5.Should_Run_Periodically_Blink_LED.fail0.save"
!!!!! Log saved to "/Users/spulverail/Documents/TechnoConcept/internship/nvp-renode//logs/Renesas_RA6M5.Should_Run_Periodically_Blink_LED.fail0.log"
+++++ Finished test 'Renesas_RA6M5.Should Run Periodically Blink LED' in 3049.29 seconds with status failed
      ╔═
      ║ Execution terminated by signal
      ║ 
      ║ Also teardown failed:
      ║ Several failures occurred:
      ║ 
      ║ 1) Connection to remote server broken: [Errno 61] Connection refused
      ║ 
      ║ 2) Connection to remote server broken: [Errno 61] Connection refused
      ║ 
      ║ 3) Connection to remote server broken: [Errno 61] Connection refused
      ║ 
      ║ 4) Connection to remote server broken: [Errno 61] Connection refused
      ╚═
```

I let it run for almost one hour before killing it with a Ctrl+C command.