renode/renode #772: DPI integration library causes a Null instance exception to be thrown

marcoantonio-ferrera, 2025-03-19:
### Description

The DPI integration library causes a Null instance exception to be thrown when dereferencing the runtime.controller object.

The issue arises because the runtime.controller and runtime.peripheral arrays are initialized in the Initial block, which runs at the start of the simulation. However, the generated block, in this case, a for loop, is part of the elaboration phase, which runs before the simulation. This leads to a situation where the controller and peripheral arrays have not been initialized, causing the Null instance exception.

### Expected behaviour

user should be able to build and run the sample examples with no error

### How to reproduce?

clone https://github.com/[antmicro/renode-dpi-examples](https://github.com/antmicro/renode-dpi-examples)

follow the readme provided in the repo

### Environment

Please, provide the following information:

Windows 11 - OS 
3c0ae21cb682be7d74a9dd8f50a36ec4fe7de480 - Renode version (with commit SHA)

### Additional information

[image]

### Do you plan to address this issue and file a PR?

yes i have a work around i am currently using

adrabarek, 2025-04-11:
Hi, thank you for reporting the issue. Please provide more information on how to reproduce it in order for us to look into it. Things that would be helpful:
- A minimal example of co-simulation setup that manifests the issue.
- The workaround that you mentioned that alleviates the issue for you.
- A way to run one of the examples from https://github.com/antmicro/renode-dpi-examples/ that triggers the issue (our CI tests are green, and I wasn't able to reproduce it by running the examples locally in the regular way).

marcoantonio-ferrera, 2025-04-14:
Hey Adrabarek, 

the issue happens when im following the read me page, it doesnt fail for the varilator version only the questa sim version, see bellow

[image]

my questasim version is :

[image]

**Work Around**
in file renode.sv commented out lines 42-52 
`  
// initial begin
  //   // runtime.controllers = new[RenodeToCosimCount];
  //   // foreach(runtime.controllers[i]) begin
  //   //   runtime.controllers[i] = new();
  //   // end

  //   // runtime.peripherals = new[CosimToRenodeCount];
  //   // foreach(runtime.peripherals[i]) begin
  //   //   runtime.peripherals[i] = new();
  //   // end
  // end
`

in file renode_pkg.sv
for the renode_runtime class i modified the constructor to initialize the controller and peripheral objects
`

    //constructors
    function new(int RenodeToCosimCount, int CosimToRenodeCount);
      controllers = new[RenodeToCosimCount];
      foreach(controllers[i]) begin
        controllers[i] = new();
      end

      peripherals = new[CosimToRenodeCount];
      foreach(peripherals[i]) begin
        peripherals[i] = new();
      end
    endfunction
`

in the sim.sv file
i pass the RenodeToCosimCount and CosimToRenodeCount when initializing the renode_runtime 
`
  renode_runtime runtime = new(RenodeToCosimCount,CosimToRenodeCount);

`
and i pass the RenodeToCosimCount to the renode class aswell
`
  renode #(
      .RenodeInputsCount(2),
      .RenodeToCosimCount(RenodeToCosimCount)
  ) renode (
      .runtime(runtime),
      .clk(clk_p),
      .renode_inputs(interrupts),
      .renode_outputs()
  );
`