renode/renode #784: ARMv8R/thumb: it/mrc execution?

emmedve1, 2025-04-07:
### Description

`it`/`mrc` sequence doesn't seem to execute correctly in `thumb` mode. `gdb`/stepping through the code executes correctly. (This executed fine on actual hardware). In `arm` mode things seem to be working as expected

### Expected behaviour

In the provided testcase, only `r0/1` are expected to contain `3FF`

### How to reproduce?

```c
#if 1
#include <arm_acle.h>
#else
#define __arm_mrc(cp, op1, CRn, CRm, op2) ({int Rt; asm("MRC p" # cp ", " # op1 ", %0, c" # CRn ", c" # CRm ", " # op2 : "=r" (Rt) : : "memory" ); Rt;})
#define __wfi()
#endif

static inline int
get_ICC_IAR(int line)
{
	return line == 0 ? __arm_mrc(15, 0, 12, 8, 0) : __arm_mrc(15, 0, 12, 12, 0);
}

[[gnu::target("arm")]]
void _start(void)
{
	asm("b __start");
	asm("b .");
	asm("b .");
	asm("b .");
	asm("b .");
	asm("b .");
	asm("b .");
	asm("b .");
	__builtin_unreachable();
}

static int ___start(int c)
{
	int i = get_ICC_IAR(c);

	if(i > 0) {
		asm("mov r1, %0" :: "r" (i));
	} else {
		asm("mov r2, %0" :: "r" (i));
	}

	return i;
}

[[gnu::used]]
static void __start(void)
{
	int c;

	asm("mov %0, pc" : "=r" (c));
	___start(c);
	while(1)  __wfi();
}
```

```bash
$ clang --target=armv8r-none-eabihf -mthumb -O3 -nostartfiles -nostdlib -Wl,-Ttext,0x8000000 -o r.elf r.c && llvm-objdump -d r.elf
r.elf:  file format elf32-littlearm

Disassembly of section .text:

08000000 <_start>:
 8000000: ea00000d      b       0x800003c <__ARMv7ABSLongThunk___start> @ imm = #0x34
 8000004: eafffffe      b       0x8000004 <_start+0x4>  @ imm = #-0x8
 8000008: eafffffe      b       0x8000008 <_start+0x8>  @ imm = #-0x8
 800000c: eafffffe      b       0x800000c <_start+0xc>  @ imm = #-0x8
 8000010: eafffffe      b       0x8000010 <_start+0x10> @ imm = #-0x8
 8000014: eafffffe      b       0x8000014 <_start+0x14> @ imm = #-0x8
 8000018: eafffffe      b       0x8000018 <_start+0x18> @ imm = #-0x8
 800001c: eafffffe      b       0x800001c <_start+0x1c> @ imm = #-0x8

08000020 <__start>:
 8000020: 4678          mov     r0, pc
 8000022: 2800          cmp     r0, #0x0
 8000024: bf14          ite     ne
 8000026: ee1c 0f1c     mrcne   p15, #0x0, r0, c12, c12, #0x0
 800002a: ee1c 0f18     mrceq   p15, #0x0, r0, c12, c8, #0x0
 800002e: 2801          cmp     r0, #0x1
 8000030: db01          blt     0x8000036 <__start+0x16> @ imm = #0x2
 8000032: 4601          mov     r1, r0
 8000034: e000          b       0x8000038 <__start+0x18> @ imm = #0x0
 8000036: 4602          mov     r2, r0
 8000038: bf30          wfi
 800003a: e7fd          b       0x8000038 <__start+0x18> @ imm = #-0x6

0800003c <__ARMv7ABSLongThunk___start>:
 800003c: e300c021      movw    r12, #0x21
 8000040: e340c800      movt    r12, #0x800
 8000044: e12fff1c      bx      r12
```

```bash
renode --console --disable-gui -e '$bin=r.elf' -e "include @scripts/single-node/cortex-r52.resc" -e start -e 'cpu GetRegistersValues' -e quit
18:29:04.1664 [INFO] Loaded monitor commands from: C:\Users\emedve\upstream\renode\scripts/monitor.py
Renode, version 1.15.3.2024 (a690bcb2-202504050107)

(monitor) $bin=r.elf; include @scripts/single-node/cortex-r52.resc; start; cpu GetRegistersValues; quit
18:29:04.2164 [INFO] Including script: C:\Users\emedve\upstream\renode\scripts\single-node\cortex-r52.resc
18:29:04.3066 [INFO] System bus created.
18:29:05.3988 [INFO] sysbus: Loading block of 244 bytes length at 0x10000.
18:29:05.4089 [INFO] sysbus: Loading block of 72 bytes length at 0x8000000.
18:29:05.4089 [INFO] sysbus: Loading block of 24 bytes length at 0x8010048.
18:29:05.4290 [INFO] cpu: Setting PC value to 0x8000000.
Starting emulation...
18:29:05.4390 [INFO] ARM Cortex-R52: Machine started.
-------------------------------
| Name   | Index | Value      |
-------------------------------
| X0     | 0     | 0x0        |
| X1     | 1     | 0x0        |
| X2     | 2     | 0x0        |
| X3     | 3     | 0x0        |
| X4     | 4     | 0x0        |
| X5     | 5     | 0x0        |
| X6     | 6     | 0x0        |
| X7     | 7     | 0x0        |
| X8     | 8     | 0x0        |
| X9     | 9     | 0x0        |
| X10    | 10    | 0x0        |
| X11    | 11    | 0x0        |
| X12    | 12    | 0x0        |
| X13    | 13    | 0x0        |
| X14    | 14    | 0x0        |
| X15    | 15    | 0x0        |
| X16    | 16    | 0x0        |
| X17    | 17    | 0x0        |
| X18    | 18    | 0x0        |
| X19    | 19    | 0x0        |
| X20    | 20    | 0x0        |
| X21    | 21    | 0x0        |
| X22    | 22    | 0x0        |
| X23    | 23    | 0x0        |
| X24    | 24    | 0x0        |
| X25    | 25    | 0x0        |
| X26    | 26    | 0x0        |
| X27    | 27    | 0x0        |
| X28    | 28    | 0x0        |
| X29    | 29    | 0x0        |
| X30    | 30    | 0x0        |
| SP     | 31    | 0x0        |
| PC     | 32    | 0x800003A  |
| PSTATE | 33    | 0x200001C0 |
| FPSR   | 66    | 0x0        |
| FPCR   | 67    | 0x0        |
| R0     | 100   | 0x3FF      |
| R1     | 101   | 0x3FF      |
| R2     | 102   | 0x3FF      |
| R3     | 103   | 0x0        |
| R4     | 104   | 0x0        |
| R5     | 105   | 0x0        |
| R6     | 106   | 0x0        |
| R7     | 107   | 0x0        |
| R8     | 108   | 0x0        |
| R9     | 109   | 0x0        |
| R10    | 110   | 0x0        |
| R11    | 111   | 0x0        |
| R12    | 112   | 0x8000021  |
| R13    | 113   | 0x0        |
| R14    | 114   | 0x0        |
| R15    | 115   | 0x800003A  |
| CPSR   | 125   | 0x200001FA |
-------------------------------
Renode is quitting
18:29:05.8701 [INFO] ARM Cortex-R52: Machine paused.
18:29:05.8802 [INFO] ARM Cortex-R52: Disposed.
```

### Environment

- Windows 11
- Renode v1.15.3.2024
  build: a690bcb2-202504050107
  build type: Release
  runtime: .NET Framework 4.0.30319.42000

### Additional information

N/A

### Do you plan to address this issue and file a PR?

N/A

emmedve1, 2025-04-07:
cc: @kknox

tomasz-worobiec, 2025-09-01:
The bug resulted from incorrect transfer of IT state between translation blocks. This should be solved by https://github.com/antmicro/tlib/commit/d7e070ee0ee43a428a6dd0e027070baf225b1a05, closing this issue.

closed 2025-09-01 by tomasz-worobiec