renode/renode #792: Unable to launch hello_world example from renode zephyr dashboard

Stanislas-MARTIN, 2025-05-06:
### Description

I am trying to simulate the hello_world example from zephyr on the mimxrt1170_evk_b_mimxrt1176_cm7 following [this](https://designer.antmicro.com/hardware/devices/mimxrt1170_evk?software=zephyr%7Cmimxrt1170_evk_b_mimxrt1176_cm7%7Chello_world)

### Expected behaviour

I am expecting to see what is shown on the website. But instead, I get the following error
```
16:23:48.4817 [INFO] Loaded monitor commands from: /home/smartin/.config/renode/renode-run.download/mono-portable/renode-1.15.3+20250505git67643848c/scripts/monitor.py
16:23:48.9952 [INFO] Including script: /tmp/tmpxgp8qti_
16:23:49.0168 [INFO] System bus created.
16:23:51.3958 [INFO] sysbus: Loaded SVD: /tmp/renode-53350/bf6339c4-3048-4976-a883-a0b34de07776.tmp. Name: MIMXRT1176_cm4. Description: MIMXRT1176DVMAA.
16:23:51.5144 [INFO] sysbus: Loading block of 40080 bytes length at 0x30000000.
16:23:51.5298 [INFO] sysbus: Loading block of 64 bytes length at 0x30009C90.
16:23:51.5299 [INFO] sysbus: Loading block of 4 bytes length at 0x30009CD0.
16:23:51.5299 [INFO] sysbus: Loading block of 4352 bytes length at 0x80000040.
16:23:54.6954 [INFO] cpu0: Guessing VectorTableOffset value to be 0x30000000.
16:23:54.7008 [ERROR] cpu0: PC does not lay in memory or PC and SP are equal to zero. CPU was halted.
16:23:54.7049 [INFO] mimxrt1170_evk_b_mimxrt1176_cm7: Machine started.
16:23:58.2274 [INFO] mimxrt1170_evk_b_mimxrt1176_cm7: Machine paused.
16:23:58.2440 [INFO] mimxrt1170_evk_b_mimxrt1176_cm7: Disposed.
```

### How to reproduce?

Follow the instruction on the previously given website or just run `renode-run demo -b mimxrt1170_evk_b_mimxrt1176_cm7 hello_world`

### Environment

Please, provide the following information:

- OS : WSL (Ubuntu 24.04 LTS)
- Renode version (with commit SHA) : 67643848c

### Additional information

N/A

### Do you plan to address this issue and file a PR?

No

pkoscik, 2025-05-13:
Hi @Stanislas-MARTIN, thanks for the report.

We have identified the issue as a limitation in the in the `renode-run` utility, we are working on resolving this.

Stanislas-MARTIN, 2025-05-16:
I encountered the same issue with the `mimxrt1062_fmurt6`, `frdm_rw612`.

pkoscik, 2025-05-16:
Hi! We have just pushed a fix to the https://github.com/antmicro/renode-run tool. Please upgrade to the latest version and let us know if this resolves your issue.

Stanislas-MARTIN, 2025-05-16:
Yes it's working now. 
Also I got a question about why there is no file system test on the dashboard ?