renode/renode #796: UART1 Echo Back Not Working Properly on Arduino Nano 33 BLE Sense in Renode

CodeCrescendo, 2025-05-27:
### Description

I am testing UART1 (Serial1) functionality using an echo-back firmware on the Arduino Nano 33 BLE Sense (based on nRF52840) in Renode. I observe that only the first byte sent to UART1 is processed correctly , further bytes are ignored or not received.

I have validated this with multiple toolchains (Arduino IDE, Simulink)
The same test works on physical hardware.

### How to reproduce?

Arduino Sketch:

```
void setup() {
  Serial1.begin(115200);
}

void loop() {
  if (Serial1.available()) {
    char c = Serial1.read();
    Serial1.write(c);
  }
}

```
my .resc script includes this:

```
emulation CreateServerSocketTerminal 12345 "term" false
connector Connect sysbus.uart0 term
```

### Environment

- OS - Windows 
- Renode version - 1.15.3.22387

Setup:

Board: Arduino Nano 33 BLE Sense (nrf52840)

ELF used: compiled either via Arduino IDE or Simulink, both give same result

Tools: MATLAB tcpclient or PuTTY

### Additional information

Observed Behavior:

- On connecting to the TCP socket (via tcpclient or PuTTY), the first character sent is echoed back as expected.

- Subsequent characters sent via the same socket are not received or echoed back.

- UART1 terminal in Renode (showAnalyzer sysbus.uart1) also shows only the first character being handled.

- This behavior occurs consistently with both Arduino-generated and Simulink-generated ELFs.

Expected Behavior:

- The sketch should echo back every character sent to UART1.

- UART1 should maintain a live two-way communication session via TCP socket

PavanKumarPothula, 2026-02-20:
I am facing the same issue. Some help would be greatly appreciated. Thanks!

jpsilvainesctec, 2026-10-21:
Hello,

I'm experiencing the same issue with Arduino. I was wondering if you managed to find a solution to this bug when using Arduino?

JPS