renode/renode #800: Unable to create a Tap

Stanislas-MARTIN, 2025-06-12:
### Description

I am following the documentation to connect to an host network but when I do `emulation CreateTap "tap0" "tap"` I get the following error
```
/usr/lib/Infrastructure.dll does not exist
  at System.IO.File.Copy (System.String sourceFileName, System.String destFileName, System.Boolean overwrite) [0x000b6] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at Antmicro.Renode.TAPHelper.DynamicModuleSpawner.GetTAPHelper () [0x0003e] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.HostInterfaces.Network.LinuxTapInterface.Init () [0x00097] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.HostInterfaces.Network.LinuxTapInterface..ctor (System.String name, System.Boolean persistent) [0x00055] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.HostInterfaces.Network.TapExtensions.CreateAndGetTap (Antmicro.Renode.Core.Emulation emulation, System.String hostInterfaceName, System.String name, System.Boolean persistent) [0x00000] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.HostInterfaces.Network.TapExtensions.CreateTap (Antmicro.Renode.Core.Emulation emulation, System.String hostInterfaceName, System.String name, System.Boolean persistent) [0x00000] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at (wrapper dynamic-method) System.Object.CallSite.Target(System.Runtime.CompilerServices.Closure,System.Runtime.CompilerServices.CallSite,object,object,object,object,object)
  at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid5[T0,T1,T2,T3,T4] (System.Runtime.CompilerServices.CallSite site, T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4) [0x0013a] in <6057d80645ae427baa7af324a4639384>:0 
  at (wrapper delegate-invoke) System.Action`6[System.Runtime.CompilerServices.CallSite,System.Object,System.Object,System.Object,System.Object,System.Object].invoke_void_T1_T2_T3_T4_T5_T6(System.Runtime.CompilerServices.CallSite,object,object,object,object,object)
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction (System.Runtime.CompilerServices.CallSite& callsite, System.Type binderType, System.Int32 knownType, Dynamitey.Internal.Optimization.InvokeHelper+LazyBinder binder, Dynamitey.InvokeMemberName name, System.Boolean staticContext, System.Type context, System.String[] argNames, System.Object target, System.Object[] args) [0x00169] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite (System.Object target, Dynamitey.InvokeMemberName name, System.Object[] args, System.String[] tArgNames, System.Type tContext, System.Boolean tStaticContext, System.Runtime.CompilerServices.CallSite& callSite) [0x0004f] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Dynamic.InvokeMemberAction (System.Object target, Dynamitey.String_OR_InvokeMemberName name, System.Object[] args) [0x00018] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext (Dynamitey.InvokeContext context, System.Reflection.MethodInfo method, System.Object[] parameters) [0x00023] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeExtensionMethod (System.Object device, System.Reflection.MethodInfo method, System.Collections.Generic.List`1[T] parameters) [0x0002c] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction (System.String name, System.Object device, System.Collections.Generic.IEnumerable`1[T] p) [0x002e9] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction (System.Type deviceType, System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x0002b] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName (System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00035] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00165] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <c40662afaaa54723be8802d84e80a768>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00aa1] in <c40662afaaa54723be8802d84e80a768>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <c40662afaaa54723be8802d84e80a768>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <c40662afaaa54723be8802d84e80a768>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x00102] in <c40662afaaa54723be8802d84e80a768>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <8e0e03f2ba714dd2a3e07c9c81e78133>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00025] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ThreadHelper.ThreadStart (System.Object obj) [0x0000f] in <aeea2e33ea654cfabae63b60ba0e127a>:0 

```

### Expected behaviour

### How to reproduce?

Launch the following command inside renode `emulation CreateTap "tap0" "tap"`

### Environment

- OS :  WSL Ubuntu 24.02
- Renode version : 
```
Renode v1.15.3.6111
  build: a690bcb2-202504080323
  build type: Release
  runtime: Mono 4.0.30319.42000
```

### Additional information

### Do you plan to address this issue and file a PR?

No

Stanislas-MARTIN, 2025-06-26:
@WojMel

Stanislas-MARTIN, 2025-06-26:
I installed it with the following link https://github.com/renode/renode/releases/download/v1.15.3/renode-1.15.3.linux-portable.tar.gz

Stanislas-MARTIN, 2025-07-16:
I will try to compile from source but there different instruction on the readme and documentation. Which one do I use ?

Stanislas-MARTIN, 2025-07-16:
I also tested the 1.15.2 version of the mono and it worked

mllofriu, 2025-08-08:
Had the same problem, running renode with sudo seems to fix it.