renode/renode #802: monitor/python variables visibility

emmedve1, 2025-06-20:
### Description

Variables visibility in `python` commands seems to favor variables **typed** in the monitor

### Usage example

`.resc` variable vs variable from keyboard:

```powershell
$ bash ~/upstream/renode/renode --disable-gui --console scripts/single-node/cortex-r52.resc
18:27:24.9743 [INFO] Loaded monitor commands from: C:\Users\emedve\upstream\renode\scripts/monitor.py
Renode, version 1.15.3.19647 (08e83a23-202506181054)

(monitor) i $CWD/scripts/single-node/cortex-r52.resc
18:27:25.2847 [INFO] Including script: C:\Users\emedve\upstream\renode\scripts\single-node\cortex-r52.resc
18:27:25.3847 [INFO] System bus created.
18:27:26.4487 [INFO] sysbus: Loading block of 25472 bytes length at 0x0.
18:27:26.4687 [INFO] sysbus: Loading block of 22520 bytes length at 0x6380.
18:27:26.4687 [INFO] sysbus: Loading block of 50 bytes length at 0xBB78.
18:27:26.4887 [INFO] cpu: Setting PC value to 0x2694.
(ARM Cortex-R52) echo $bin
C:\Users\emedve\AppData\Local\Temp\renode-24360\e31e4957-79e8-40ac-a227-789960b476fd.tmp
(ARM Cortex-R52) python "print(variables['bin'])"
None
(ARM Cortex-R52) python "print(monitor.GetVariable('bin'))"
None
(ARM Cortex-R52) $bin=@file.name
(ARM Cortex-R52) python "print(variables['bin'])"
file.name
(ARM Cortex-R52) python "print(monitor.GetVariable('bin'))"
file.name
(ARM Cortex-R52)
```

Command line variables vs variables from the keyboard:

```bash
$ bash ~/upstream/renode/renode --disable-gui --console -e '$file=@file.name' -e 'i @scripts/single-node/cortex-r52.resc'
18:36:20.5791 [INFO] Loaded monitor commands from: C:\Users\emedve\upstream\renode\scripts/monitor.py
Renode, version 1.15.3.19647 (08e83a23-202506181054)

(monitor) $file=@file.name; i @scripts/single-node/cortex-r52.resc
18:36:20.7683 [INFO] Including script: C:\Users\emedve\upstream\renode\scripts\single-node\cortex-r52.resc
18:36:20.8638 [INFO] System bus created.
18:36:21.9022 [INFO] sysbus: Loading block of 25472 bytes length at 0x0.
18:36:21.9180 [INFO] sysbus: Loading block of 22520 bytes length at 0x6380.
18:36:21.9180 [INFO] sysbus: Loading block of 50 bytes length at 0xBB78.
18:36:21.9338 [INFO] cpu: Setting PC value to 0x2694.
(ARM Cortex-R52) echo $file
file.name
(ARM Cortex-R52) python "print(variables['file'])"
None
(ARM Cortex-R52) python "print(monitor.GetVariable('file'))"
None
(ARM Cortex-R52) $file=@other_file.name
(ARM Cortex-R52) python "print(variables['file'])"
other_file.name
(ARM Cortex-R52) python "print(monitor.GetVariable('file'))"
other_file.name
(ARM Cortex-R52)
```

### Additional information

Version and host platform available in the logs above

### Do you plan to address this issue and file a PR?

Would much appreciate an initial analysis of the examples?

WojMel, 2025-06-23:
Hi @emmedve1,

The `GetVariable` method doesn't fallback to global scope as it's done for variable expansion in the Monitor shell. Until it's changed you can use `variables['global.bin']` to specify the context explicitly.

emmedve1, 2025-06-23:
This works for me. Thank you @WojMel :-)

closed 2025-08-20 by p-woj