renode/renode #803: linux-arm64 binary builds?

jmgrosen, 2025-06-27:
Thanks so much for adding arm64 host support! Your blog post announcing it says you support it on both Linux and macOS, but it appears that the linux-arm64 builds on builds.renode.io aren't being maintained--that is, all other platforms have builds within the last month, but the latest linux-arm64 build is from May 13th. Are you planning to continue to build newer versions of linux-arm64?

LillyJinstrand, 2025-06-30:
Hi @jmgrosen ,
Uploading of the arm64 linux packages was disabled due to an issue with the CI, but it seems it was not re-enabled when that issue was fixed. That should be corrected now, so tomorrows nightly builds should include linux-arm64 again

LillyJinstrand, 2025-07-01:
New nightly builds are now up, and the up to date package can once again be found at https://builds.renode.io/renode-latest.linux-arm64-portable-dotnet.tar.gz

closed 2025-07-01 by LillyJinstrand