renode/renode #806: execution tracing causing crash

ky-macpherson, 2025-07-03:
Thanks for taking the time to file an issue for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.md) first.

### Description

I have a m33 code sequence that is causing renode to crash when execution tracing is enabled.

### Expected behaviour

When execution tracing is not enabled, the test runs successfully.

### How to reproduce?

Will provide a link

### Environment

- OS = Ubuntu 22.04.5 LTS
- up to date master sha = 2c5a60e6d890d866656b3932d5be15a89c12918c

### Additional information

n/a

### Do you plan to address this issue and file a PR?

I do not currently have a fix.

ky-macpherson, 2025-07-03:
https://github.com/renode/renode-issue-reproduction-template/blob/3ec5f29aef3072b62473792b696d8f6a6bfb28b0/test.robot

MateuszKarlic, 2025-07-04:
As of: https://github.com/renode/renode/commit/d2c941ba3d467cfbbbbfc4c2d9a3149208cf4c6f this is fixed

closed 2025-07-04 by MateuszKarlic